bittide/bittide-hardware #165: Shake inconsistent rebuild behaviour due to relative pathing

lmbollen, 2022-11-07:
While adding components to `bittide-instances` I noticed that shake's rebuild behaviour depends on your current path.
The rebuild behavior depends on the `dist-newstyle` directory in your current path, while it `dist-newstyle` is actually located in the project folder (where `cabal.project` is located).

We should change the implementation to locate the `cabal.project` file and depend on the `dist-newstyle` in that directory.

lmbollen, 2022-11-07:
@martijnbastiaan PTAL

martijnbastiaan, 2022-11-07:
Agreed, this is an issue. These lines need to change:

https://github.com/bittide/bittide-hardware/blob/4b57c30bacfdfd678f45e87ba651c31f90ed8bf6/bittide-instances/bin/Shake.hs#L112

https://github.com/bittide/bittide-hardware/blob/4b57c30bacfdfd678f45e87ba651c31f90ed8bf6/bittide-instances/bin/Shake.hs#L26-L27

commit df413b18 (2022-11-21): Use absolute build dirs in Shake files

Fixes #165
PR #183 (2022-11-21): Use absolute build dirs in Shake files
Fixes #165
commit 66a1e49c (2022-11-21): Use absolute build dirs in Shake files (#183)

Fixes #165
closed 2022-11-21 by martijnbastiaan