bittide/bittide-hardware #209: Add boot stage 0: configure clock boards

martijnbastiaan, 2022-12-14:
Should:

1. Write config over SPI
2. Wait for clock to become stable
3. Deassert reset for boot stage 1

Make sure to add a "Boot" section to the README describing this boot stage.

lmbollen, 2022-12-21:
Updates:
*We have an implementation that performs the mentioned steps, what is left is to split them up in separate PRs.

lmbollen, 2022-12-22:
#216 contains an implementation that performs these steps.
The image below shows the SPI communication between the FPGA and the `Si5395J` chip once it is plugged in.
[image]
Observe following:
* All SPI transactions in the beginning are the FPGA trying to read a register that indicates that the chip is ready for communication.
* Once the chip is ready, the FPGA writes the preamble, after that there's a long pause.
The long pause is a ~300 ms wait state required after writing the preamble, required by the datasheet. 
* After the long pause the configuration is written, followed by a short pause.
The short pause is the time it takes for the FPGA's PLL to acquire lock on the generated clock.  About 24 ms.
* After the short pause, frequency control over SPI is enabled, callisto runs at an update rate of 10kHz, each bump takes about 1.3us, followed by a 98.8 us pause.

martijnbastiaan, 2023-07-01:
Fixed in https://github.com/bittide/bittide-hardware/pull/215

closed 2023-07-01 by martijnbastiaan