bittide/bittide-hardware #239: Investigate accidental double-ack in `DoubleBufferedRam`

martijnbastiaan, 2023-01-25:
During the integration of VexRisc (https://github.com/bittide/bittide-hardware/pull/198) we found that `DoubleBufferedRam` misbehaves when requesting multiple reads back-to-back. It would behave as follows:

[image]

I.e., it would ACK the request for `a2` while presenting data corresponding to `a1` on the bus. The component should have provided backpressure as follows:

[image]

With Contranomy this didn't used to be an issue, as every request would look like:

[image]

I.e., Contranomy would simply ignore the wrongly presented data because it wouldn't read back-to-back.

-----

The PR introducing VexRisc piggy-backs a fix for this issue, but IMO this should have been caught by tests. Let's investigate:

 * Why this wasn't caught by tests
 * What we can do to prevent similar issues in the future

martijnbastiaan, 2023-01-25:
Related: https://github.com/bittide/bittide-hardware/pull/198#discussion_r1082542395

cuddlefishie, 2023-01-27:
By the wishbone specification technically outputting two requests in the "same" CYC+STB phase is allowed, and a slave is also allowed to keep asserting ACK. The specification uses this "multiple requests in a row" as an example to introduce registered feedback, so it seems *discouraged* but legal.

The `driveStandard` function in `clash-protocols` always deasserts CYC after a transaction, so any test using this would not have caught this error.

I also noticed that the `validateLenient` circuit sometimes steps "too quickly" when updating the validation state. I will fix this.

I will also add an option to the `driveStandard` function so that the number of cycles where CYC+STB are deasserted can be configured, that would allow it to also test situations where they are never deasserted.

cuddlefishie, 2023-02-14:
Changes to clash-protocols are implemented in https://github.com/clash-lang/clash-protocols/pull/52 and the changed/added tests for bittide are implemented in #251

closed 2023-04-19 by martijnbastiaan