bittide/bittide-hardware #291: Reduce Rust binary sizes

martijnbastiaan, 2023-05-03:


cuddlefishie, 2023-05-03:
A utility program that repackages ELFs and only keeps what we are interested in can be found here: https://github.com/cuddlefishie/elf_repack.

We probably want to not actually transfer full ELFs, because all `#![no_std]` ELF readers do not support streaming, so a custom streamable format will be implemented.
This makes reduction in program size more important than gains from reducing the size of the ELF container.

cuddlefishie, 2023-05-03:
Things to investigate:

- [ ] using [`ufmt`](https://lib.rs/crates/ufmt)
- [ ] building with `build_std` to remove panic messages in `core` and optimise for size

cuddlefishie, 2023-05-09:
Results from using `ufmt` and `build_std` on the example firmware binaries

Note: these numbers are only "fixed" memory usage, the `.stack` section that grows to fill all available data memory is excluded.

To make displaying and *checking* those sizes easier, there's now [cuddlefishie/elf-limits](https://github.com/cuddlefishie/elf-limits), which was used to get those numbers below.

## `hello`

Configuration | Total memory | Instructions | Data
-- | --:| --:| --:
Baseline | 24.28 KiB | 18.05 KiB | 6.23 KiB
`build_std` | 12.04 KiB | 9.32 KiB | 2.73 KiB
`build_std` + `ufmt` | 5.73 KiB | 5.31 KiB | 428 B

## `fdt-read`

Configuration | Total memory | Instructions | Data
-- | --:| --:| --:
Baseline | 29.13 KiB | 22.29 KiB | 6.84 KiB
`build_std` | 15.38 KiB | 12.34 KiB | 3.04 KiB
`build_std` + `ufmt` | 7.66 KiB | 7.26 KiB | 404 B

closed 2023-05-24 by martijnbastiaan