bittide/bittide-hardware #296: Find out why Questa is slow in combination with Clash-FFI

martijnbastiaan, 2023-05-10:
- [ ] Is cocotb slow in similar situations? If not, what are the differences?

kleinreact, 2023-07-04:
The problem seems to be the stack of dynamically linked Haskell libraries, which results from a `Cabal` based build process by default. Questa seems to encounter some heavy performance issues when loading all these dynamic resources (at least under Linux). The problem goes away if all the Haskell dependencies are statically linked into the shared library instead, which is, unfortuantely, an intricate process to set up.

---

The following steps / resources are required to produce a Haskell based VPI library, which statically links all the Haskell dependencies directly into the resulting shared library:

* Some Linux distro that is built `-fPIC` with support. We have chosen Debian 12 (bookworm), as our tests with Ubuntu based images did not succeed.
* Some recent GHC and Cabal executable. We chose GHC 9.0.2 and Cabal 3.6 (the default on Debian 12 is Cabal 3.4, resulting in strange dependency issues. We got rid of the error by updating Cabal to 3.6).
* GHC must be patched to build static libraries with `-fPIC` by default (including the base libraries shipped with GHC itself). We applied the following patchbefore building GHC:
  ```patch
  --- compiler/GHC/Driver/Session.hs
  +++ compiler/GHC/Driver/Session.hs
  @@ -3723,6 +3723,7 @@ default_PIC platform =
       -- while we could work around this on x86_64 (like WINE does), we won't be
       -- able on aarch64, where this is enforced.
       (OSDarwin,  ArchX86_64)  -> [Opt_PIC]
  +    (OSLinux,   ArchX86_64)  -> [Opt_PIC, Opt_ExternalDynamicRefs]
       -- For AArch64, we need to always have PIC enabled.  The relocation model
       -- on AArch64 does not permit arbitrary relocations.  Under ASLR, we can't
       -- control much how far apart symbols are in memory for our in-memory static
  @@ -4623,7 +4624,7 @@ picCCOpts dflags = pieOpts ++ picOpts
             ["-fPIC", "-U__PIC__", "-D__PIC__"]
         -- gcc may be configured to have PIC on by default, let's be
         -- explicit here, see #15847
  -       | otherwise -> ["-fno-PIC"]
  +       | otherwise -> ["-fPIC"]
  ```
  Furthermore, the following `build.mk` has been used:
  
  ```
  SRC_HC_OPTS          += -fPIC
  SRC_CC_OPTS          += -fPIC
  EXTRA_HC_OPTS         = -fPIC
  GhcStage1HcOpts       = -fasm -O0
  GhcStage2HcOpts       = -fasm -O0
  GhcLibHcOpts          = -fasm -O2
  GhcLibWays            = v dyn
  DYNAMIC_GHC_PROGRAMS  = YES
  DYNAMIC_BY_DEFAULT    = NO
  SplitObjs             = NO
  HADDOCK_DOCS          = NO
  BUILD_DOCBOOK_HTML    = NO
  BUILD_DOCBOOK_PS      = NO
  BUILD_DOCBOOK_PDF     = NO
  V                     = 1
  LATEX_DOCS            = NO
  HSCOLOUR_SRCS         = NO
  BeConservative        = YES
  ```
* GHC should be configured to use the system `libffi` instead of the one shipped with GHC. Our target test environment will be on `libffi8`, which means we cannot use `ghcup` which still depends on `libffi7`. Debian 12 is on `libffi8` already including the required Cabal.
* Profiling should be disabled and shared libraries should be enabled by default.

All of these constraints can be satisfied using the following `Dockerfile`:
```Dockerfile
FROM debian:bullseye

RUN apt-get update -y \
 && apt-get install -y \
        locales wget curl git cabal-install \
        build-essential \
 && sed -i -e \
        's/# en_US.UTF-8 UTF-8/en_US.UTF-8 UTF-8/' \
        /etc/locale.gen \
 && locale-gen \
 && echo "deb http://downloads.haskell.org/debian bullseye main" >> /etc/apt/sources.list \
 && apt-key adv --keyserver keyserver.ubuntu.com --recv-keys BA3CBA3FFE22B574 \
 && apt-get update -y \
 && apt-get install -y cabal-install-3.6 \
 && cd /usr/local/bin \
 && ls /opt/cabal/bin/ | xargs -I{} ln -s /opt/cabal/bin/{}
[... 20 lines cut ...]
 && make install \
 && rm -rf /compile /my-build.mk /fpic-default.patch \
 && cd /usr/local/bin \
 && ls /ghc-9.0.2-fpic/bin/ | xargs -I{} ln -s /ghc-9.0.2-fpic/bin/{}
```

Now, with the right GHC and Cabal executable at hand, we can build our library. Make sure to start on a fresh environment, as all of your Haskell dependencies have to be rebuild.

We aim to build a `foreign-library` that can be loaded into Questa, as it is done [here](https://github.com/clash-lang/clash-compiler/blob/ad42dc5f92f7cee6529e8e2b7e8db8555840d5c1/clash-testbench/example/clash-testbench-example.cabal#L49). However, in that setup all Haskell dependencies are dynamically linked. To link them statically, we now can add a `ghc-options: -static` to our `.cabal`-file, which only works because of all the previous preparations. This should get rid of all the dynamically linked Haskell dependencies except for the `rts`. We don't know why Cabal refuses to statically link the `rts`. There is also the `options: standalone` setting for foreign libraries to archive exactly that, but Cabal also refused to apply that one in our tests.

So Cabal always links the `rts` dynamically by default and there does not seem to be an option to turn that off. Our hacky workaround is to build the `foreign-library` with `ghc-options: -v`, grab the last linker call and just remove the `-ghc9.0.2` (this suffix is always added to the dynamic libraries) from the `rts` linking argument and re-execute that call. Then, all the Haskell based dependencies are builtin.

The resulting library now finally is free of dynamically linked Haskell dependencies, but, _bad news_, it won't load into Questa. The reason is that all of the shared FFI entry points are gone. They usually are offered by `cl [... cut ...]
_Note that not all of the above constraints may be necessary to get this running. Some further tests are required to simplify these steps._

Finally, many thanks to the authors of the following resources, which helped a lot for putting all of this together:
* https://www.hobson.space/posts/haskell-foreign-library
* https://github.com/unclechu/haskell-experiment-shared-library-for-c-application/tree/master/docker/ghc-with-fpic
* https://gitlab.haskell.org/ghc/ghc/-/issues/17618
* https://mail.haskell.org/pipermail/ghc-devs/2017-December/015245.html

martijnbastiaan, 2023-07-05:
Impressive debugging work!

Could we do something similar to [`mkblob`](https://github.com/sigurd-dev/mkblob)? Maybe we apply do some trickery that would make Questa un-responsible for loading shared libraries?

martijnbastiaan, 2023-07-25:
> Could we do something similar to [mkblob](https://github.com/sigurd-dev/mkblob)? Maybe we apply do some trickery that would make Questa un-responsible for loading shared libraries?

No. 

Alright, so this seems pretty much investigated now. Hopefully we'll ever find a proper way around it..

closed 2023-07-25 by martijnbastiaan