bittide/bittide-hardware #439: Perform parameter sweep on hardware and gather statistics

martijnbastiaan, 2023-11-15:
We'd like to do a bunch of experiments to get a good feeling for clock control behavior in actual hardware. Ideally we'd sweep over a number of parameters:

 - Clock offsets
 - Topologies

To do this we need a number of things:

 - [ ] A way to do clock offsets in hardware
 - [ ] Topology configuration in hardware
 - [ ] A way to quickly execute a number of tests with different parameters

lmbollen, 2024-01-04:
### Current HITL infrastructure
Set up test environment and check if required probes are present.
We currently assume that every FPGA contains the same bitstream and only check if the first FPGA has the required probes.
We collect the names of the start probes.
Since we can have multiple tests per bitstream we can have multiple start probes.
- For each start probe name:
  - For each device:
    - Initialize the device
    - Arm trigger and capture for each ILA
    - Set start probe
  - For each device:
    - Load & refresh the device
    - Wait for the test to end
    - Print the result
  - For each device:
    - Load & refresh the device
    - Load and store ILA data
  - Print the test summary

Finally, we print the summary of all tests.
Each test has a start, done and success probe.
### Proposed changes
#### Test stage control
We can still use a start probe for every independent test, but we can also do different tests in different stages. 

To create different stages in our test infrasctructure we add two probes: `num_stages` and `current_stage`.
Change the `done` probe and `success` probe into `stage_done`and `stage_fail`.
`num_stages` is controlled by the FPGA and represents the number of stages in the current test.
`current_stage` is controlled by Vivado and incremented every time `stage_done` is true.
If `current_stage == num_stages && stage_done` the test has succeeded.
#### Extra probe control
Each test can have a yaml file that contains stage information. In each stage we can assign new values to additional probe files, such as:
```yaml
stages:
  - name: reset
    index: 0
    probes:
    - devices:
      - index: 0
        link_mask: 0b0000_0001
        period_offset: 100
      - index: 1
        link_mask: 0b0000_0010
        period_offset: -100
  - name: setup
    index: 1
  - name: synchronize
    index: 2
  - name: apply period changes
    index: 3
  - name: synchronize
    index: 4

``` 
#### ILA control
For every stage, we arm the ILAs at the start, and check them at the end.
At the end we can extract the data if the ILA has been triggered, store it in a folder that corresponds to that stage and re-arm for the next stage.

martijnbastiaan, 2024-01-05:
As discussed offline I think the stages can be handled by the FPGAs/tests themselves, like we do now.

```yaml
$start_probe_name:
  - $test_name:
    - index: $index
      config:
        $input_probe_name: $input_probe_value
```

where:

* `$start_probe_name`: refers to name of start probe
* `$test_name`: free form field, probably used to name the stored ILA data
* `$index`: FPGA to use, if left out implicitly counts from zero
* `$input_probe_name`: an input probe on the VIO with the start probe
* `$input_probe_value`: value to set input probe to before asserting start probe

I'm also wondering whether we could define this in Haskell, that way you could use native data types for the input probe values..

---------------------

Example:

```yaml
start_probe_a:
  - star:
    - index: 0
      config:
        link_mask: 0b0000_0001
        period_offset: 100
    - index: 1
      config:
        link_mask: 0b0000_0010
        period_offset: -100

# Empty dictionaries for auto inference
start_probe_b: {}
```

lmbollen, 2024-01-31:
After a discussion with Felix (who implemented infra to generate YAML files based on HITL instances). We've decided to write a more elaborate proposal regarding the complete HITL infra.
One of the goals of this write up is to isolate functionalities for maintainability and separation of concerns.

The proposed setup consists of three separate systems:

1. Hardware Test TCL for Vivado.
A generic bittide agnostic script to execute HITL tests. This script executes a test as described in a yaml file.
The main difference with the current implementation is that we move configuration variables to outside the tcl. This includes for example the `fpga_ids` and timeout configurations. The goal would be to have a uniform way of describing HITL tests that can be used for all FPGA platforms. Preferable in a human readable / writable format.

2. Designing HITL infra to generate correct configurations.
Felix has been working on infra that enables us to design tests in Haskell and generate correct YAML files. Correct meaning that the generation of the YAML configurations is based on the types used in the testbench.

3. Generating multiple tests with random data.
Where part 2 focuses on generating yaml files for trivial tests based on the test bench. This part focuses on generating tests for testbenches that have parameters. (to be extended)

kleinreact, 2024-01-31:
I'd like to add on that from the designer's perspective, who is creating and maintaining the tests.

First of all, I think there are still some mixed concepts here that need to be clarified, as we have different conceptual layers of test design:

1. __Test Instances__ (corresponding to the bitstream uploaded to each of the FPGAs)

   Currently, we use one Haskell module per instance, which gets synthesized to a separate bitstream in the HDL. Conceptually, each test instances should correspond to different functionality of the system resulting in a manageable amount of instances. Running different instances is expensive, as every instance needs to be synthesized separately and requires a new upload to the FPGAs.

2. __Test Cases__ (being part of a test instance)

   Here, the same instances is used to run some different kind of tests. This is especially useful to share resources or system state among different tests, to avoid otherwise redundant setup routines and boot procedures with respect to dedicated hardware components that just need some time to get set up. Test cases still are reasonable different in the sense that they target different control flow for their execution and don't just swap out some data value. The amount of test cases per test instance should be human manageable as well, in the sense that each test case can manually crafted and maintained.

3. __Parameters__ (of test instances / cases)

   A test case still may utilize some data constants that can be adapted within some well-defined ranges. The parameters should be chosen in such a way that the executed test case follows a similar control flow for all possible parameter variations. If a parameter change triggers some completely different test behavior, then splitting the test into multiple test cases instead should be considered, where we assume that the number of different behaviors to test is manageable (_i.e., it is very unlikely that the components we are testing obey so many completely different behaviors that we cannot handle without an automatic test coverage strategy_).

These are the different aspects we need to consider for test design. For test execution, we then fix some parameters, a test case and a test instance, which we refer to as a __Test__.

Our targeted infrastructure should be able to handle these different design aspects in a transparent fashion.

Especially, it must be clarified:
  * Which aspect of a configuration file targets which layer?
  * How does the infrastructure set up and run a particular __test__?

Another crucial points is the requirement of test case selection as part of the hardware design. As we don't want to create a new bitstream per test, we need to select the test case and need to pass the parameters to the hardware at runtime. As @lmbollen pointed out, we need to define how much the test infrastructure, especially the TCL interface, needs to be aware of this, i.e., does it utilize a particular notion of test case and parameters or not? From the view of the interface, there are three possible options:

1. The TCL infra is aware of test cases and uses multiple VIO probes to separate test selection and data distribution. That's how we currently do it in [BoardTest.hs](https://github.com/bittide/bittide-hardware/blob/staging/bittide-instances/src/Bittide/Instances/Hitl/BoardTest.hs#L122).

2. The TCL does not separate test selection and parameters, it just pass some data from the configuration file to the probe.

   1. Nevertheless, the Clash interface at the hardware side knows how to separate that data in a uniform way and takes care that test cases and parameter configuration don't get messed up (also see my proposal [here](https://github.com/bittide/bittide-hardware/blob/hitl-yaml-cfgs/bittide-instances/data/hitltconfigs/README.md) to this).

   2. There is no particular separation between test case selection and parameters at the hardware level either. In other words, the selection of a different test cases just can be considered to be a parameter as well. It is up to the test designer to handle that correctly. This's how we currently do it in [FincFdec.hs](https://github.com/bittide/bittide-hardware/blob/staging/bittide-instances/src/Bittide/Instances/Hitl/FincFdec.hs#L28).

martijnbastiaan, 2024-02-06:
All good points @kleinreact. It took a bit too long, but for now we've settled on settled on:

* Defining a technology agnostic way to define tests. See [here](https://github.com/bittide/bittide-hardware/pull/460#issuecomment-1926524529). This should be reusable in other projects / other FGPA platforms.
* We'll squash (2) and (3). That is, we'll do what you described as _"There is no particular separation between test case selection and parameters at the hardware level either. In other words, the selection of a different test cases just can be considered to be a parameter as well. It is up to the test designer to handle that correctly. This's how we currently do it in FincFdec.hs."_

I think this is the simplest thing we can do, while incing closer to a situation that's ergonomic.

closed 2024-04-25 by kleinreact