bittide/bittide-hardware #449: Reframing on HITLT

kleinreact, 2024-01-10:
Reframing currently fails on the HITL tests to successfully recenter the buffer occupancies of the elastic buffers (cf. [#3455](https://github.com/bittide/bittide-hardware/actions/runs/7445553156)).

[image]

It has to be investigated whether this is a theory, implementation, or methodology issue. 

_Note that  waiting times are sufficiently large. All of the nodes store the right correction values for reframing (captured at the beginning of the red markings indicating the waiting times)._

lmbollen, 2024-01-12:
I think it's important that we coordinate this investigation with @sanjaylall in the loop.

martijnbastiaan, 2024-01-17:
So @sanjaylall noticed in one of the meetings that these plots (as opposed to the simulation plots) don't have symmetric buffer occupancies. These should be there though, given the duplex nature of our fiber connections. It's currently very unclear what could be causing this behavior given that 1) we use the same code base for simulation and synthesis 2) the way we construct the counters is incredibly simple. More investigation is needed!

PR #453 (2024-01-24): Fix asymmetric buffer occupancies
#415 changes the ILA dump structure for the elastic buffer sizes to only report differences since the last and the current capture in order to compress the amount of data being reported. 

It may be that a difference grows too fast and exceeds the compressed data-size-boundary between two captures, which is why a fail-safe procedure was implemented creating some additional capture before that will happen (resetting the difference counter). However, this protection mechanism was not setup correctly. As a consequence, the compressor sometimes ends up in a situat [... cut]