bittide/bittide-hardware #476: CI fails due to failing license check

kleinreact, 2024-03-12:
cf. https://github.com/bittide/bittide-hardware/actions/runs/8238420951/job/22529333608

DigitalBrains1, 2024-03-12:
What a weird error. It sounds like it didn't think the `current_branch` in the script is named `staging` but it looks like that should be the case.

I wonder if the script could be simplified if this were fixed:
```shell
# XXX: We want to pass in the target branch of the PR, but we only run
#      on 'push' (see 'on' section in this YAML). Because of this, GH
#      sets the target branch to an empty string.
```
Shouldn't it be the case that the branch [is accessible as](https://docs.github.com/en/actions/learn-github-actions/contexts#github-context)  `${{ github.ref }}`?

DigitalBrains1, 2024-03-12:
I doubt if the script has ever worked correctly. The most recent nightly that succeeds is [this](https://github.com/bittide/bittide-hardware/actions/runs/7768872493) and it doesn't mention the `License year check` job at all.

martijnbastiaan, 2024-03-13:
There are two related, but not the same, issues/annoyances here:

-------

One: The script wants to check all commits from `HEAD` to the common ancestor of the merge target. If it sees that there are zero commits it concludes something is not up to snuff and throws an error. This check makes sense on a PR, because why would you want to merge an empty PR? Something must be wrong in the script logic then! However, it doesn't make sense on nightly: there is no PR, so there is nothing to check. Hence https://github.com/bittide/bittide-hardware/pull/479.

------

Two: _Ideally_ you'd be able to create a chain of PRs, each getting checked until their merge target. I.e., not `origin/staging`, but `origin/what-the-pr-is-based-on`. I figured this must have been `${{ github.ref }}`, _but_:

> The fully-formed ref of the branch or tag that triggered the workflow run. **For workflows triggered by `push`, this is the branch or tag ref that was pushed.** For workflows triggered by pull_request, this is the pull request merge branch

and note that the following:

https://github.com/bittide/bittide-hardware/blob/8e70f595d36d0c1da7d65a6df7ca5a5ddbcb2467/.github/workflows/ci.yml#L20

..means that we create a `push` event, not a `pull_request` event. That is to say, `${{ github.ref }}` will always be the branch pushed to, not the merge target. We don't want to switch to `pull_request`, because that means you always have to have a PR open any time you want to run something on CI.

martijnbastiaan, 2024-03-13:
This issue has been fixed by https://github.com/bittide/bittide-hardware/pull/479. The `XXX` will have to linger in the source code :(.

closed 2024-03-13 by martijnbastiaan

DigitalBrains1, 2024-03-13:
I completely misinterpreted what the _target branch_ was in the note I quoted, that's where I got confused.