bittide/bittide-hardware #481: Sovling the constant crisis

martijnbastiaan, 2024-03-14:
We currently hard code memory addresses all across our code base. As far as I can tell, they are scattered across three parts of our code base:

1. `clash-vexriscv`. It contains a (hard coded) reset vector (which we patch in this repo to be somewhere else, but that's another story).
2. Our hardware designs in `bittide-instances`
3. Our Rust code in `firmware-binaries`

This has always been undesirable, but lately it has been reaching a boiling point due our stack maturing and us wanting to do more and more in software. To solve this, we basically have two options:

1. Define a memory map, and integrate it in our hardware designs. This is what https://github.com/bittide/bittide-hardware/pull/315 has proposed, based on https://github.com/bittide/bittide-hardware/issues/47.

2. Use _the power of Clash_ to create a protocol agnostic wrapper for memory mapped protocols, see a proposal below.

Originally (2) was just a weekend project of mine and I wouldn't have proposed it as an alternative to (1). However, during its implementation I found that:

* This would allow us to generate Rust data types / getters / setters for the contents of the registers, straight from the Haskell types.
* This would allow us to generate documentation straight from Haskell types (this would need some work, so that's not really a MVP goal).
* It doesn't rely on Clash constant folding in presence of HO-functions. (Alternatively worded: we're not restricted in the way we design our circuits.)
* We can change ONE thing (the circuit) and have the rest follow suit. That it, there is no way the DTS would get out of sync with the design.

I'm not sure how much work there is left to be done, but it seems worthwhile to at least investigate to see how much time it would take to get to an MVP.

--------------------------------

```haskell
{-# LANGUAGE FlexibleContexts #-}
{-# LANGUAGE MagicHash #-}
{-# LANGUAGE NamedFieldPuns #-}

-- | A protocol agnostic wrapper for memory mapped protocols. See 'MemoryMapped'
-- for more information. Throughout this module we'll refer to devices initiating
-- a transaction on a bus as \"managers\" and devices receiving a transaction as
-- \"subordinates\".
--
-- TODO:
--
--   * Provide examples on how to use the data structures protocols
--   * Provide standard memory map aware registers and interconnects
--   * Provide a way to generate human readable documentation
--   * Provide a way to generate data structures for target languages
--   * Provide a way to check whether memory maps are valid
--
module Protocols.MemoryMapped where

import Control.Monad (void, forM)
import Data.Bits (popCount)
import Data.Foldable (toList, Foldable (foldl'))
import GHC.Int (Int(I#))
import GHC.Integer.Logarithms (wordLog2#)
import GHC.Stack (HasCallStack)
import GHC.Word (Word(W#))

import Protocols.Internal (Protocol(Fwd, Bwd), Circuit(..))
import Control.Monad.Extra (when)
import Data.Tuple.Extra ((&&&))
import Data.List (sortOn)
import Clash.Explicit.Prelude (NFDataX(deepErrorX), Natural)

-- | Absolute address or offset in bytes
type Address = Integer

-- | Size in bytes
type Size = Integer

-- | A protocol agnostic wrapper for memory mapped protocols. It provides a way
-- for memory mapped subordinates to propagate their memory map to the top level,
-- possibly with interconnects merging multiple memory maps.
--
-- The current implementation has a few objectives:
--
--   1. Provide a way for memory mapped peripherals to define their memory layout
[... 151 lines cut ...]
    overlaps (l1, u1) (l2, u2) = l1 < u2 && l2 < u1

-- | Calculate the size of a field in bits
fieldTypeSize :: FieldType -> Word
fieldTypeSize = \case
  BitVectorFieldType w -> w
  SignedFieldType w -> w
  UnsignedFieldType w -> w
  VecFieldType n f -> n * fieldTypeSize f
  SumOfProductFieldType cs ->
      nConstructorsBits (wordLength cs)
    + maximum [sum [fieldTypeSize (thing f) | f <- thing con] | con <- toList cs]
 where
  wordLength :: Foldable f => f a -> Word
  wordLength = fromIntegral . length

  nConstructorsBits :: HasCallStack => Word -> Word
  nConstructorsBits 0 = 0
  nConstructorsBits v@(W# w)
    | isPowerOfTwo = r
    | otherwise = r + 1
    where
      isPowerOfTwo = popCount v == 1
      r = fromIntegral (I# (wordLog2# w))
```

martijnbastiaan, 2024-03-15:
Having talked about this offline, we envision something like this possible:

```haskell
-- A circuit mapping registers:
[fooBus, barBus] <-
  [foo, bar, wibble] <- mmInterconnect @8
    (  (0x0, 4)
    :> (0x4, 4)
    :> (0x8, 4)
    :> Nil 
    ) -< mainBus

mmReg "foo"    -< foo
mmReg "bar"    -< bar
mmReg "wibble" -< wibble

-- A circuit mapping devices:
[uartBus, timerBus, wibbleBus] <-
  mmInterconnect @16
    -- addr size
    ( (0xA, sMiB 1)  -- UART
   :> (0xC, sMiB 16) -- Timer
   :> (0x8, sMiB 32) -- Wibble
   :> Nil) -< mainBus

uartWb -< uartBus
timerWb -< timerBus
wibbleWb -< wibbleBus
```

where

```

mmReg :: String -> Circuit (MemoryMapped ...) ()

mmInterconnect :: Vec n (Offset, Size) -> Circuit (MemoryMapped ...) (Vec n (MemoryMapped ...))
```

Because we don't use the `MemoryMap` for the circuit, we sidestep any issues with Clash constant folding.

bgamari, 2024-04-16:
FWIW, I have handled register space layout at the type level with seemingly good results in my (now mis-named) [axi-register](https://git.smart-cactus.org/ben/axi-register/-/blob/2.0/src/Protocols/Register.hs?ref_type=heads). Type inference is decent (although I think can be improved) and the RTL generated seems sensible.

martijnbastiaan, 2025-07-16:
Given all the work that has been done and how content we are, I think we can call this done! :D