bittide/bittide-hardware #539: Generalize `Bittide.Hitl.FpgaIndex`

kleinreact, 2024-06-16:
Currently, `bittide-experiments` defines https://github.com/bittide/bittide-hardware/blob/703b881e8e468e7916dc01b2f8c49ba559ac4649/bittide-experiments/src/Bittide/Hitl.hs#L94-L96 being fixed to work for 8 node topologies, but the number of FPGAs available may change and should be configured in `bittide-instances` instead: https://github.com/bittide/bittide-hardware/blob/703b881e8e468e7916dc01b2f8c49ba559ac4649/bittide-instances/src/Bittide/Instances/Hitl/Setup.hs#L21-L22

Therefore, it may be desirable to generalize the definition in `bittide-experiments` towards an arbitrary number of FPGAs or to implement the proposal given in the comment.

_Note that `bittide-experiments` cannot just use the value from `bittide-instances`, as this would introduce circular dependencies._

martijnbastiaan, 2024-06-16:
This might be a little bit much to fall into the scope of this particular issue, but I wonder how easy it would be to read a specific test setup from configuration files. E.g., CI would have a setup with 8 nodes, while smaller setups would only have 3. Of course, not all topologies would be testable on a 3-setup, but the code generating test cases could account for that. Most other tests (tranceiver up, riscv test, etc.) shouldn't have to care all that much.

kleinreact, 2024-06-17:
Check out the changes introduced by [mini-bittide](https://github.com/bittide/bittide-hardware/tree/mini-bittide), where you can see what is necessary for changing the current setup to work with `1 FPGA`/`2 Nodes`. It shouldn't be that much effort to generalize that to work towards `n FPGAs`/`m Nodes` setups as well.

JvWesterveld, 2024-09-09:
In #536, bittide-experiments no longer hard codes the number of FPGAs in the HITL setup.

martijnbastiaan, 2024-09-11:
Yeah, I think #536 fixes Felix's original issue. If we feel like it we could open another issue for https://github.com/bittide/bittide-hardware/issues/539#issuecomment-2171821675, but until we've got multiple setups (again) I think that's low priority. Closing, thanks @JvWesterveld! :)

closed 2024-09-11 by martijnbastiaan