bittide/bittide-hardware #541: Add `Clash.Sized.Byte`

martijnbastiaan, 2024-06-17:
Clash offers `BitVector`, but in memory mapped systems we often use bytes. To this end we've introduced type synonyms:

https://github.com/bittide/bittide-hardware/blob/703b881e8e468e7916dc01b2f8c49ba559ac4649/bittide/src/Bittide/SharedTypes.hs#L46-L49

It would be nice if we would get a whole bunch of convenience functions that work on bytes, e.g. `shiftL`.

DigitalBrains1, 2024-06-17:
We should make a class `ClashBits` (?) where bit indices and counts are expressed as `Index n` (for `n`-wide words) instead of the default `Bits` class with its `Int`s. Those are pretty annoying; right now Jasper is using `fromEnum` to go from `Index n` to `Int` without getting the dreaded warning about `fromInteger` during HDL generation.

martijnbastiaan, 2024-06-18:
Yeah or just `Bits`, and make a `Clash.Compat.Prelude18` for backwards compat. (Though it must be noted this is the bittide repo :-))