bittide/bittide-hardware #549: Add `gild` as a formatter

martijnbastiaan, 2024-06-17:
https://taylor.fausak.me/2024/02/17/gild/

lmbollen, 2024-07-29:
I took a look at both `cabal-gild` and `cabal-fmt` and concluded that these formatters would severely impact the clarity of our cabal files simply because they do not handle comments nicely.

We use comments to explain why we add certain flags or enable pragmas, but these comments are moved to another location.

Below you can find the diffs introduced by the different tools:
`cabal-gild`
```diff
--- /home/lucas/bittide-hardware/bittide/bittide.cabal
+++ /home/lucas/bittide-hardware/bittide/bittide.cabal.fmt
@@ -1,19 +1,26 @@
-cabal-version:       2.4
-name:                bittide
-synopsis:            Bittide hardware descriptions, which should be
-                     target independent, loosely connected / standalone
-version:             0.1
-License:             Apache-2.0
-license-file:        LICENSE
-author:              QBayLogic B.V.
-maintainer:          devops@qbaylogic.com
-Copyright:           Copyright © 2022-2024 Google LLC
-
+cabal-version: 2.4
+name: bittide
+synopsis:
+  Bittide hardware descriptions, which should be
+  target independent, loosely connected / standalone
+
+version: 0.1
+license: Apache-2.0
+license-file: LICENSE
+author: QBayLogic B.V.
+maintainer: devops@qbaylogic.com
+copyright: Copyright © 2022-2024 Google LLC
 data-files:
   data/clock_configs/*.csv
 
 custom-setup
-  setup-depends: Cabal, base, directory, process, filepath, extra
+  setup-depends:
+    Cabal,
+    base,
+    directory,
+    extra,
+    filepath,
+    process,
 
 -- You probably want add a `cabal.project.local` that looks like:
 --
@@ -25,11 +32,15 @@
     Enable the Rust FFI for the Callisto clock control. Disable this if you want
     working HLS for the bittide package. Obviously, any tests using the Callisto
     Rust implementation will fail.
+
   default: True
   manual: True
 
 common common-options
   default-extensions:
+    -- TemplateHaskell is used to support convenience functions such as
+    -- 'listToVecTH' and 'bLit'.
+    -- Prelude isn't imported by default as Clash offers Clash.Prelude
     BangPatterns
     BinaryLiterals
     ConstraintKinds
@@ -46,53 +57,51 @@
     InstanceSigs
     KindSignatures
     LambdaCase
+    NoImplicitPrelude
     NoStarIsType
[... 462 lines cut ...]
-    tasty-hunit,
-    tasty-th,
+
+  build-depends:
+    , base
+    , bittide
+    , bittide-extra
+    , bytestring
+    , clash-cores
+    , clash-lib
+    , clash-prelude
+    , clash-prelude-hedgehog  >=1.6    && <1.10
+    , clash-protocols
+    , constraints             >=0.13.3 && <0.15
+    , containers
+    , elf
+    , extra
+    , hedgehog                >=1.0    && <1.5
+    , HUnit
+    , tasty                   >=1.4    && <1.6
+    , tasty-expected-failure
+    , tasty-hedgehog          >=1.2    && <1.5
+    , tasty-hunit
+    , tasty-th
```

lmbollen, 2024-07-29:
We chose to add the explained flags to the comment blocks.
This will be fixed by #583

PR #583 (2024-07-31): Add `cabal-gild`
Fixes #549
closed 2024-07-31 by martijnbastiaan