bittide/bittide-hardware #587: Implement fast/efficient execution deadlines in software

martijnbastiaan, 2024-07-31:
We currently maintain two instances for clock control: hardware and software. We use the former because it both consistently and reliably hits a (maximum) clock control update frequency of 1 MHz. While the latter does what we want in practice (clock convergence), we don't have a good idea on how long it takes to do a single clock control update. I.e.:

[image]

We don't really mind fewer updates (say, 100 KHz) but the updates should be timed reliably to make sure we can make our theory match our practice. We currently _[... out of time, will finish this issue later ...]_

leonschoorl, 2024-08-29:
I added some [measurement hardware](https://github.com/bittide/bittide-hardware/commit/9927aaebcbf373c78094679b1083d11883415ffb) and ran [a test](https://github.com/bittide/bittide-hardware/actions/runs/10600359274/job/29416763045).

With the current setup the period between clock updates fluctuates between 405 and 414 cycles @ 125MHz.
Except for some reason on node 2, where it fluctuates between 403 and 412 cycles.
So the update frequency is `125 MHz / 414 = 0.301 MHz` or 301 kHz.
And the period is `414 / 125 MHz = 3.31e-6 s` or 3.31 μs

rslawson, 2024-11-08:
I didn't realise that this would get closed when even just one of the linked PRs got closed, oops.

rslawson, 2024-11-27:
Oops, still not done with #654.

closed 2024-12-16 by rslawson