bittide/bittide-hardware #597: Clock reports should use relative measures (`PPM`), not absolute ones (`fs`)

martijnbastiaan, 2024-08-12:
We currently produce plots like these:

[image]

On the Y-axis we find a measure called "relative period (fs)". What this means is that the observed clocks have a clock period of `clock_period_of_measurement_clock` + `relative_period`. So in the figure above we can see a clock at `-400 fs`. Though never mentioned in the report, the measuring clock is running at 125 MHz or a clock period of `8 ns` ~ `8_000_000 fs`. And so, the clocks differ 0,005% or 50 ppm.

I've long found this a very counterintuitive measure, and have gotten several comments on its difficulty to parse from others too. I think the reason it feels counterintuitive stems from two things:

1. Other data sheets - when speaking of frequencies - mention relative measures such as PPM (or PPB). Think of oscillator inaccuracies (SkyWorks), FINC/FDEC step sizes (SkyWorks), frequency offset (Xilinx).
2. Just the relative period tells us nothing of how close we are to the desired frequency. An offset of 1 picosecond on a 10 MHz clock is nothing to worry about, but on a 10 GHz its pretty bad!

Partly due to reason (1), but mostly due to (2) I think we should switch to PPM.

hiddemoll, 2024-08-13:
The clock reports also lists 'clock offsets (fs)' in the summary at the top of each page. This should perhaps also be changed to ppm/ppb.

DigitalBrains1, 2024-08-13:
What about a plain ratio? So 1 ppm = 1e-6, 1 ppb = 1e-9 (short scale). These ppm/ppb things aren't the prettiest constructs.

martijnbastiaan, 2024-08-14:
IMO it just makes it harder to have a conversation about the plots. Given that short form is the default in English (if not straight definition) makes me think we can safely use it.

commit 39d3c50a (2024-08-14): Use PPM as measure in clock plots

Fixes #597
commit 9968084b (2024-08-14): Use PPM as measure in clock plots

Fixes #597
closed 2024-08-14 by martijnbastiaan