bittide/bittide-hardware #609: Remove unit juggling in clock test pipeline

martijnbastiaan, 2024-08-20:
We currently juggle a few units that all amount to the same thing:

* Number of FINC/FDEC presses
* PPM/PPB/PPT
* Femtoseconds
* Number of clock ticks

Worse still, these often are encoded in the same type. This has lead to a bunch of different interpretation issues and bugs that have gone unnoticed for months. As defended [here](https://github.com/bittide/bittide-hardware/issues/597) I think we should use PPM (or a divisor) as our ground unit, and only convert on the "borders" of our design where necessary.

leonschoorl, 2024-08-22:
Within (parts of) the hardware, it makes sense to natively used "number of FINC/FDEC steps". That way we don't have to bother with fractional(or worse, floating) numbers, and the associated cost/complexity.

And to avoid accidentally mixing up these different things, we should separate them with (new)types.

martijnbastiaan, 2024-08-26:
https://github.com/bittide/bittide-hardware/pull/618 uses PPx (`PartsPer`) as a "core" type and it does clean up nicely.

commit c5eff308 (2024-08-26): Directly compare expected and measured n.o. cycles

Instead of taking a roundtrip to absolute time calculations, we directly
compare the expected number of cycles assuming an ideal frequency, with
the measured number of cycles. This removes almost all type juggling.

Fixes #609
commit 0d76f155 (2024-08-26): Directly compare expected and measured n.o. cycles

Instead of taking a roundtrip to absolute time calculations, we directly
compare the expected number of cycles assuming an ideal frequency, with
the measured number of cycles. This removes almost all type juggling.

Fixes #609
commit 9055791d (2024-08-26): Directly compare expected and measured n.o. cycles

Instead of taking a roundtrip to absolute time calculations, we directly
compare the expected number of cycles assuming an ideal frequency, with
the measured number of cycles. This removes almost all type juggling.

Fixes #609
commit 1cd1eca6 (2024-08-26): Directly compare expected and measured n.o. cycles

Instead of taking a roundtrip to absolute time calculations, we directly
compare the expected number of cycles assuming an ideal frequency, with
the measured number of cycles. This removes almost all type juggling.

Fixes #609