bittide/bittide-hardware #611: Investigate "discreteness" of clock plots

martijnbastiaan, 2024-08-21:
If we look at any clock plot that starts of sufficiently closely to the convergence frequency, we can observe a certain "blockiness" or "discreteness" of the plots:

[image]

The step size corresponds to roughly 0.5 ppm, while the clock step size is 0.01 ppm (10 ppb). I don't think this is fundamental to our measurement method:

* We sample every 5 ms, or every 5×125_000_000 = 625_000_000 (ideal) clock cycles.
* 0.5 ppm is therefore 312.5 clock cycles
* 10 ppb is therefore 6.25 clock cycles

While I can see there might be some synchronization logic messing with the precision around ~10 ppb, I don't think 0.5 ppm should be a cut-off point. Perhaps a good first step would be to remove all the type juggling in https://github.com/bittide/bittide-hardware/issues/609 to get a good view on where all the conversions happen.

martijnbastiaan, 2024-08-26:
Oh no, another case of _I'm off by 3 orders of magnitude_. The correct logic would be:

 * We sample every 20 ms, or every 20×125_000 = 2_500_000 (ideal) clock cycles, given a clock of 125 MHz.
 * 0.5 ppm is therefore 2.5 clock cycles
 * 10 ppb is therefore 0.025 clock cycles

And so, because the source data dances around -1,0,1 we _do_ expect "glitches" of ~0.5 ppm.

closed 2024-08-26 by martijnbastiaan

hiddemoll, 2025-01-21:
Leaving this comment for future us.

To quote Martijn from Slack:
> The plot shows measured frequency vs "guessed" frequency. The former through measuring frequency "directly" by measuring how many cycles an fpga counted too many/too few compared to a reference clock. The latter by plotting where an fpga thinks it is based on the number of times it pressed FINC/FDEC.

[image]

Martijn wondered why we only see 3 states in the measured frequency (black line); one measuring artifact up, zero, or down. In the previous comment Martijn says:
> We have two clock crossings, so we can expect ~4 clock cycles of uncertainty, or 1.6 ppm.

But it is actually 3 clock cycles of uncertainty. The clock domain crossings can fall either fall left-left, right-right, left-right or right-left, but the last two states have the same clock cycle uncertainty.

martijnbastiaan, 2025-01-21:
Hmm, in other plots I _do_ see 4 values:

[image]

So I'm not entirely sure about the methodology (either way).