bittide/bittide-hardware #648: Add haskell functions to interact with `gdb`

lmbollen, 2024-10-09:
We currently have some functionality to start `gdb` and run a script that is located outside of the Haskell code. But it would be nice to at least have a reusable function to load a binary onto the FPGA.
This could simply be a function that starts openocd, starts gdb, programs the soft-core connected via JTAG and starts execution.

martijnbastiaan, 2024-11-01:
Perhaps we could pass these in using environment variables? I.e., something like:

```
USB_DEVICE="1-5.1:1" gdb the_script.tcl
```

lmbollen, 2024-11-01:
@martijnbastiaan How would we then open multiple jtag devices at once?

martijnbastiaan, 2024-11-01:
We're currently using [`withCreateProcess`](https://hackage.haskell.org/package/process-1.6.25.0/docs/System-Process.html#v:withCreateProcess), which takes a [`CreateProcess`](https://hackage.haskell.org/package/process-1.6.25.0/docs/System-Process.html#t:CreateProcess). Each `CreateProcess` can pass its own environment variables ([`env`](https://hackage.haskell.org/package/process-1.6.25.0/docs/System-Process.html#v:env)). I.e., you would start OpenOCD multiple times, each with its own environment variable.

lmbollen, 2024-11-01:
Ah interesting. Currently @hiddemoll has a working implementation, but that seems like a way simpler approach. You'll see the PR soon I think

commit 0c0e757b (2024-11-05): Use environment variable to open OpenOCD with configurable adapter USB location

Fixes: #648
commit ecc19c1d (2024-11-06): Use environment variable to open OpenOCD with configurable adapter USB location

Fixes: #648
closed 2024-11-06 by hiddemoll