bittide/bittide-hardware #691: Rust Backtraces?

rslawson, 2024-12-04:
In PR #690 I introduced a GDB script that should be able to grab backtraces when Rust code panics in the TCP test. However, despite my best efforts, those backtraces are limited to a depth of 2, and one of the displayed functions is invariably `core::panicking::panic` (or related). I'm not sure if this is because something is ignoring `panic = "abort"`, or if the `-C force-unwind-tables=yes -C force-frame-pointer=yes` options are being ignored somehow, but this is definitely worth investigating.

martijnbastiaan, 2025-02-06:
Tracking in: https://github.com/rust-lang/rust/issues/136198.

rslawson, 2025-04-30:
Small update: [my PR on `riscv-rt`](https://github.com/rust-embedded/riscv/pull/254) has been merged, which in concert with moving to nightly Rust should allow more experimentation with getting backtraces. Previously we couldn't get representative builds because we had to disable LTO, but now that should no longer be necessary.