bittide/bittide-hardware #702: Possibly use `core::intrinsics::volatile_copy_nonoverlapping_memory` to address throughput concerns

rslawson, 2025-01-09:
In the current implementations of the `ScatterUnit` and `GatherUnit` HALs, the memory copying is implemented manually with `for` loops in order to avoid the compiler optimising out MMIO accesses. However there exists `core::intrinsics::volatile_copy_nonoverlapping_memory`, which is an equivalent to the `llvm.memcpy.p0i8.0i8.*` intrinsic with the `volatile` parameter set to `true`, so it will not be optimised out unless the size is equal to zero. In practice this compiles to `tail memcpy`, which is exactly what the prior (and faster) implementation which used `copy_to_nonoverlapping` compiles to as well. I'm not completely certain that this will actually result in performance gains, and it also requires a nightly compiler to be able to enable `#![feature(core_intrinsics)]`, so this will likely require some extra work in infra to enable that.

rslawson, 2025-01-09:
Here's a [Godbolt link](https://godbolt.org/z/hcoovW89c) showing the Assembly output I was viewing.

rslawson, 2025-01-13:
Never mind, after talking it over with some embedded devs in Rust groups it's been concluded that `copy_to_nonoverlapping` is sufficient for our use cases, since the order of writes to/from the slice doesn't matter, just that they all happen.

closed 2025-01-13 by rslawson

hydrolarus, 2025-01-13:
Does `copy_to_nonoverlapping` guarantee that the read does happen though? What if there's two reads after another (when data changes underneath through MMIO)? I can't find such a guarantee in the docs?