bittide/bittide-hardware #738: Add DDR4 primitive

martijnbastiaan, 2025-03-31:
We'd like to store debug information (and potentially CC program memory) in DDR to save on block rams. To do that we need a wrapper for DDR (see: PG150) and configure it with the right parameters for our board.

Notes: 

 * Parameters can be hard coded.
 * See https://github.com/clash-lang/clash-compiler/pull/2343 for reference implementation.
 * See Xilinx example project for constraints / test code.

What I roughly expect:

```haskell
ddr4 ::
  -- Routed directly from top:
  "sys_clk"          ::: DiffClock sysDom ->
  "c0_ddr4_clk"      ::: Signal sysDom (BitVector 17) ->
  "c0_ddr4_cke"      ::: Signal sysDom (BitVector 17) ->
  "c0_ddr4_bg"       ::: Signal sysDom (BitVector 2) ->
  "c0_ddr4_ba"       ::: Signal sysDom (BitVector 2) ->
  "c0_ddr4_adr"      ::: Signal sysDom Bit ->
  "c0_ddr4_odt"      ::: Signal sysDom Bit ->
  "c0_ddr4_cs_n"     ::: Signal sysDom Bit ->
  "c0_ddr4_ck_t"     ::: Signal sysDom Bit ->
  "c0_ddr4_ck_c"     ::: Signal sysDom Bit ->
  "c0_ddr4_reset_n"  ::: Signal sysDom Bit ->
  "c0_ddr4_dm_dbi_n" ::: BiSignalIn 'SFloating sysDom 1 ->
  "c0_ddr4_dq"       ::: BiSignalIn 'SFloating sysDom 8 ->
  "c0_ddr4_dqs_t"    ::: BiSignalIn 'SFloating sysDom 1 ->
  "c0_ddr4_dqs_c"    ::: BiSignalIn 'SFloating sysDom 1 ->

  -- Routed from our own designs:
  "axi4_m2s"         ::: Signal uiDom Axi4M2S ->

  ( -- Note that the "_o" signals don't show up in HDL. Their "BiSignalIn" parts
    -- do though.

    -- Routed directly to top:
    "c0_ddr4_dm_dbi_n_o" ::: BiSignalOut 'SFloating sysDom 1
  , "c0_ddr4_dq_o"       ::: BiSignalOut 'SFloating sysDom 8
  , "c0_ddr4_dqs_t_o"    ::: BiSignalOut 'SFloating sysDom 1
  , "c0_ddr4_dqs_c_o"    ::: BiSignalOut 'SFloating sysDom 1
  , "axi_s2m"            ::: BiSignalOut 'SFloating sysDom 1

    -- Routed to our own designs:
    Clock uiDom ->
    Signal uiDom Axi4S2M
  )
```

martijnbastiaan, 2025-04-28:
@hiddemoll I've published https://github.com/clash-lang/clash-cores/pull/30 which gets you part of what you need for this issue.

hiddemoll, 2025-04-29:
I've published [clash-lang/clash-cores#31](https://github.com/clash-lang/clash-cores/pull/31) which adds `DiffClock` support to `inst`, which we need for this issue.

closed 2025-06-24 by hiddemoll