bittide/bittide-hardware #779: Investigate suspicious instance `GBitPackCSums U1`

martijnbastiaan, 2025-05-21:
In https://github.com/bittide/bittide-hardware/pull/777 I add:

```haskell
{- | XXX: Though the number of constructors a unit constructor is zero (I think?)
we have to set it to at least 1 due to the @1 <=@ constraint on the class.
-}
instance GBitPackCSums U1 where
  type GNumConstructors U1 = 1
  type GSumSize U1 = 1
  type GSumAlignment U1 = 1

  gConstructorTag n _ = n
  gpackCsum U1 = 0
  gunpackCsum _ _ = U1
```

I'm not entirely sure about the value of `GSumSize`: it seems that the size is not introduced by this unit constructor itself, rather than all the constructors of the data type. At the same time I don't see how this introduces issues looking at the use sites, plus, I'm forced by the class constraints to write this.

@hydrolarus opinions?

hydrolarus, 2025-05-21:
What was the code that caused this to be an issue again? I am currently reworking the `BitPackC` type class so I could take a look and make sure in the new version it doesn't pop up either

martijnbastiaan, 2025-05-21:
`shake vexRiscvTest:hdl` should trigger it. It's on Clash's normalization side.

commit 5abf02d3 (2025-06-03): Rework `BitPackC` to correctly do padding and byte ordering

Fixes #797
Fixes #779
commit 4b052171 (2025-06-04): Rework `BitPackC` to correctly do padding and byte ordering

Fixes #797
Fixes #779
commit 66b21d0f (2025-06-04): Rework `BitPackC` to correctly do padding and byte ordering

Fixes #797
Fixes #779
commit e866a6a5 (2025-06-05): Rework `BitPackC` to correctly do padding and byte ordering

Fixes #797
Fixes #779
closed 2025-06-06 by martijnbastiaan