bittide/bittide-hardware #781: `BitPackC (Unsigned n)` doesn't pack in a useful way

martijnbastiaan, 2025-05-21:
In Haskell:

```haskell
  initWb2 :: Unsigned 32
  initWb2 = 0xDEAD_ABBA
```

Printing in Rust:

```
baabadde
```

The implementation for `Signed` _does_ look correct. The implementation for `Double` and `Float` (at least one of them) is broken too AFAICT.

martijnbastiaan, 2025-05-21:
This does what I want:

```diff
diff --git a/clash-protocols-memmap/src/BitPackC.hs b/clash-protocols-memmap/src/BitPackC.hs
index f6d0d611..ffea7093 100644
--- a/clash-protocols-memmap/src/BitPackC.hs
+++ b/clash-protocols-memmap/src/BitPackC.hs
@@ -88,8 +88,8 @@ instance (KnownNat n) => BitPackC (Unsigned n) where
   type ByteSizeC (Unsigned n) = NextPowerOfTwo (SizeInBytes n)
   type AlignmentC (Unsigned n) = ByteSizeC (Unsigned n)
 
-  packC val = packC (bitCoerce val :: BitVector n)
-  unpackC bits = bitCoerce (unpackC bits :: BitVector n)
+  packC val = toLittleEndian $ packC (pack val :: BitVector n)
+  unpackC bits = unpack (unpackC (fromLittleEndian bits) :: BitVector n)
 
 instance (KnownNat n) => BitPackC (Signed n) where
   type ByteSizeC (Signed n) = NextPowerOfTwo (SizeInBytes n)
@@ -114,15 +114,15 @@ instance BitPackC Float where
   type ByteSizeC Float = 4
   type AlignmentC Float = 4
 
-  packC = packC . pack
-  unpackC = unpack . unpackC
+  packC = toLittleEndian . packC . pack
+  unpackC = unpack . unpackC . fromLittleEndian
 
 instance BitPackC Double where
   type ByteSizeC Double = 8
   type AlignmentC Double = 8
 
-  packC = packC . pack
-  unpackC = unpack . unpackC
+  packC = toLittleEndian . packC . pack
+  unpackC = unpack . unpackC . fromLittleEndian
 
 instance BitPackC Bool where
   type ByteSizeC Bool = 1
```

Though this is a stupid implementation: it undoes earlier endiannes transformations..

martijnbastiaan, 2025-05-22:
I'll also say that this endianness stuff is frying my brain a bit, so the ticket above is purely reasoned from a position of: I'm feeding a result of `packC` to the data bus and when I print it over UART I get flipped bytes.

hydrolarus, 2025-05-23:
Can we repurpose this issue to investigate all of endian handling in bittide maybe? I feel like these instances are only symptoms

hydrolarus, 2025-06-06:
Closed with #801

closed 2025-06-06 by hydrolarus