bittide/bittide-hardware #787: Add `reset` value back to memory map

martijnbastiaan, 2025-05-23:
After merging https://github.com/bittide/bittide-hardware/pull/786, the Rust side won't parse the the reset value of registers anymore. As the TODO states:

```rust
    // TODO: There is a field called reset in the JSON, but we don't use it yet.
    //       It is commented out for now, because:
    //
    //         1. Its type is wrong: registers can be much larger than u64
    //         2. The JSON parser interprets "large" numbers (correctly) as
    //            floating point numbers, which is not what we want.
    //
    // pub reset: Option<u64>,
```

hydrolarus, 2025-06-16:
What if the field was an array of numbers (bytes, specifically)? Then we could use the BitPackC instances and equivalent to give this data some meaning?

martijnbastiaan, 2025-06-16:
That would work!