bittide/bittide-hardware #832: Extend `registerWbC` and friends with atomic operations

martijnbastiaan, 2025-06-09:
Take the following type:

```haskell
type T = Maybe (Unsigned 32)
```

This gets encoded as:

```
CC......UUUUUUUU
```

where `C` is the constructor byte, and `UUUUUUUU` the `Unsigned 32`. For example, if we'd write a `Just 0xBADCOFFE`, we'd write:

```
01......BADCOFFE
```

Because we're operating on a 32-bit bus, we'd first write the constructor bits, followed by the unsigned (or the other way around). The problem with that is that the hardware will see any of the intermediate representations. E.g. when transitioning from a `Nothing`:

```
00......00000000
```

it might see:

```
01......00000000
```

or `Just 0`. This is clearly undesirable: the hardware should only see _before_ or _after_ the full write operation. I think we've got a couple of options:

1. Do nothing. If users of `registerWbC` want atomic operations they'd have to implement it themselves.
2. Add a single "transaction" register to every device, where `data Transaction = Begin | Abort | Commit`. After writing `Begin` to this register, no bus write would take effect, until `Commit` has been written. Similarly, `Abort` just discards whatever has been written so far. The HAL would automatically wrap any write to a "composite" type or an "atomic" type that is larger than the bus width in a transaction. (Of course, with optional escape hatches.)
3. Same as (2), but add a transaction register for every register that holds a "composite" type or an "atomic" type that is larger than the bus width in a transaction.

The first option is not great IMO. I think the fact that small composite types (e.g. `Maybe Word8`) still get written in multiple transactions will continue to bite us. The second option would add an additional register per register, which might be a bit wasteful. Still, it would auto-add the ability for users to write many registers atomically, which might be cool! The third one saves some space over the second one and does away with the "global" transaction feature.

Opinions @hydrolarus?

hydrolarus, 2025-06-10:
I agree, it will probably come to bite us at some point.

Both (2) and (3) are fine, if we want to lock the whole device at every register access then (2) is good. At the moment there's no handling of interrupts or multithreading or something like this, so for now this approach should work.

I'm not aware of any designs on how we want to handle interrupts or if we ever want to go multithreaded, so I'd say going with (2) for now is fine?

martijnbastiaan, 2025-06-10:
I think (2) is reasonably simple to implement, so let's go with that then. I've got some vague ideas on multithreading/interrupts, but I think we're pretty far from that at the moment so let's just cross that bridge when we get there :).

lmbollen, 2025-06-11:
I don't see any problem with (1) tbh. If you want atomic operations that's _very_ easy to achieve with a single `registerWbC`

leonschoorl, 2025-06-19:
@martijnbastiaan In passing I mentioned the [A extension](https://www2.eecs.berkeley.edu/Pubs/TechRpts/2016/EECS-2016-118.pdf#%5B%7B%22num%22%3A131%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C72%2C724.245%2Cnull%5D), but after some more careful reading, I now think that wouldn't actually help.
It adds some atomic operations, but only on machine words.

If do some kind of locking/transaction system like (2) or (3):
 * In theory it's probably important to do proper [FENCE](https://www2.eecs.berkeley.edu/Pubs/TechRpts/2016/EECS-2016-118.pdf#%5B%7B%22num%22%3A102%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C72%2C621.069%2Cnull%5D)-ing to make sure a `Commit` doesn't get re-ordered with a (partial) data write.
   Although in practice it probably won't matter, as I don't think the VexRiscv does any reordering.
  * More importantly if you do decide to implement (2) or (3), make sure to also think carefully about opposite direction: values being written from the hardware 
    The issue seems to talking from the perspective of the (CPU) bus master updating a register, and the hardware seeing partially updated values.
    But the hardware could also update the contents of the register. And although that always happens atomically, it could happen that the bus master is in the middle of a read operation, having read some, but not all the `wordSize`d chunks of a register, and then the hardware side updates the whole values.
   Then the CPU will end up having read part of the old and part of the new value.

leonschoorl, 2025-06-26:
Instead of the explicit synchronization described in 2 (and 3), we could consider making the synchronisation implicit.

For a group of addresses (of whatever size makes sense to synchronise together), we could:
 * on write: redirect writes to a set of shadow registers, and only once the last one gets written to, trigger an atomic copy of the shadow registers to the actual registers
 * on read: have reading of the first address trigger an atomic copy of the actual registers to the shadow registers, and then read from the shadow registers

This kind of approach probably makes sense when you go towards smaller grained synchronisation.
As it could save a lot of address space "clutter", and extra writes.
But you'd have to take care that the program doesn't take shortcuts, like only writing the word containing the tag bit of a `Nothing` of some wide Maybe type.

martijnbastiaan, 2025-07-07:
> More importantly if you do decide to implement (2) or (3), make sure to also think carefully about opposite direction: values being written from the hardware

That's a good point, I hadn't considered that.

The implicit path sounds too dangerous too me, exactly for the reason you point out: if the Rust compiler ever decides to make that `Nothing`-optimization (which is IMO perfectly acceptable), everything would break.