bittide/bittide-hardware #904: HAL generation crashes on zero-width registers

martijnbastiaan, 2025-07-22:
For giggles (or not to be a _schijtluis_ as @lmbollen phrased it) I tried to use the memory map in combination with zero-width registers. First off, this didn't work because of https://github.com/bittide/bittide-hardware/issues/903. I then made an ad-hoc change to make it report to the memory map that the construct takes 1 word, which then crashes the generator:

```
error: failed to run custom build command for `bittide-hal v0.1.0 (/home/martijn/code/bittide-hardware/firmware-support/bittide-hal)`

Caused by:
  process didn't exit successfully: `/home/martijn/code/bittide-hardware/firmware-binaries/../_build/cargo/firmware-binaries/debug/build/bittide-hal-df4ec732368fd1b8/build-script-build` (exit status: 101)
  --- stdout
  cargo::rerun-if-changed=/home/martijn/code/bittide-hardware/firmware-support/bittide-hal/../../_build/memory_maps

  --- stderr

  thread 'main' panicked at /home/martijn/code/bittide-hardware/firmware-support/memmap-generate/src/generators/mod.rs:44:5:
  Ident is not allowed to be empty; use Option<Ident>
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
firmware-support failure!
firmware-binaries failure!
```

hydrolarus, 2025-07-22:
The issue lies in the caller of `zeroWidthRegisterMeta` and that not giving a name to the register.

The error message could be improved, here's a small patch that just adds a check for empty identifiers.

```diff
diff --git a/firmware-support/memmap-generate/src/generators/mod.rs b/firmware-support/memmap-generate/src/generators/mod.rs
index 9774c79a..5dfc9c62 100644
--- a/firmware-support/memmap-generate/src/generators/mod.rs
+++ b/firmware-support/memmap-generate/src/generators/mod.rs
@@ -41,6 +41,10 @@ pub fn ident(ident_type: IdentType, n: impl AsRef<str>) -> Ident {
         IdentType::Raw => s.to_string(),
     };
 
+    if s.is_empty() {
+        panic!("Name passed to `generators::ident({ident_type:?})` is empty")
+    }
+
     Ident::new(&s, Span::call_site())
 }
 

```

martijnbastiaan, 2025-08-06:
Fixed in https://github.com/bittide/bittide-hardware/pull/929

closed 2025-08-06 by martijnbastiaan