bittide/bittide-hardware #950: Special case `Maybe` in Rust code generator?

martijnbastiaan, 2025-08-08:
I have a type in Haskell:

```haskell
data CallistoConfig = CallistoConfig
  { waitTime :: Maybe Word32
  ...
```

Which gets converted to:

```rust
pub struct CallistoConfig {
    pub wait_time: Maybe<u32>,
```

It seems to me that it would have been more Rustean to generate:

```rust
pub struct CallistoConfig {
    pub wait_time: Option<u32>,
```

Would it be an option to "hijack" types like these to generate more Rustean code?

hydrolarus, 2025-08-08:
Yes, this is also something I want! With the current `ToFieldType` rewrite I'm planning to add better support for special casing types, also things like `()` and `Either` would be on the list! So I'm on it already kinda 😅

hydrolarus, 2025-12-04:
About this, I don't think we can do this without making things much more complex. `Option` has the `#[repr(Rust)]` which means it can optimise the layout in certain cases, we don't want that, so I don't think we can do this 😔 

Tuples are done now, but I'm actually not sure if they use C repr, so maaaybe that might come back to bite us, but for now it seems to work 👀

martijnbastiaan, 2025-12-04:
In retrospect I don't think it's a good idea anyway. Developers will write documentation based on the Haskell types, which will get transferred to the Rust docs. This is going to get confusing. 

Do you agree?